SunPower/PVMismatch #39: replace pvcell with abstract IV-curve API

mikofski, 2017-01-12:
replace pvcell with an IV curve backend API that can be changed to another backend for either 5-parameter model or a "pvsyst" like model.

bmeyers, 2017-02-07:
I was just testing the [`onediode`](https://pvlib-python.readthedocs.io/en/latest/generated/pvlib.pvsystem.singlediode.html#pvlib.pvsystem.singlediode) method in `pvlib.pvsystem`, and it doesn't appear to have any reverse bias behavior modeled:

[image]

It seems like we unfortunately can't drop that in as a 1:1 replacement for `.calcCell`.

mikofski, 2017-02-07:
Unfortunately no, but I wasn't expecting that they would have reverse bias because it's not part of the DeSoto or 5-parameter model. However it shouldn't be a problem to drop the forward bias part into pvcell and leave the reverse bias alone:

    Icell = Iforward + I reverse

bmeyers, 2017-02-07:
I see, that makes sense