SunPower/PVMismatch #40: tips and tricks section

mikofski, 2017-01-12:
There are a few tricks that aren't obvious, so should there be a tips section of the docs?

For example, it is usually faster to create a new `PVsystem` then change every single cell in the system to a new irradiance.

```python
>>> s=time.time();\
...     pvm=pvmodule.PVmodule(pvcells=pvcell.PVcell(Ee=0.88));\
...     pvsys=pvsystem.PVsystem(pvmods=pvm);\
...     print time.time()-s
# 0.523000001907 seconds

>>> s=time.time();\
...     pvsys.setSuns(0.75);\
...     print time.time()-s
# 1.37000012398 seconds
```

For example if you are trying to do a full day simulation of a shadow passing over the array, then to set the ambient effective irradiance on the cells at each timestep, you should call `pvsys = pvsystem.PVsystem()` at each timestep instead of `pvsys.setSuns()`.

bmeyers, 2017-01-12:
I definitely think so. I actually hadn't even thought of this particular trick. I'll try to keep other tips in mind as I'm continuing to work on ShadeResearch.

mikofski, 2017-01-12:
I think we can collect tips on the [wiki](https://github.com/SunPower/PVMismatch/wiki#tips-and-tricks) and transfer the best tricks to the docs with each release.