SunPower/PVMismatch #41: serialization and de-serialization methods for module, string and system objects

mikofski, 2017-01-18:
a generalized method that converts an object to a string or binary form that can be passed to and from mapping methods.

mikofski, 2017-01-18:
This can also be used to save the system state.

wholmgren, 2017-01-18:
I want this for pvlib objects too.

bmeyers, 2017-02-08:
One issue I came across when implementing this for ShadeResearch was the need to define things at the cell, module, and system level. This assumes that everything is uniform in the state you want to save, and that all you need to capture are:

-  The diode-equation parameters
- The number of points per IV curve
- The module stringing configuration (e.g., `cell_pos`)
- Cell area
- System stringing configuration (number of modules and strings)

 I solved this by making three dictionaries: one for each set of kwargs for each class. I then reconstruct the system from within the mapping function like so:

```
pvcell = PVcell(**pvcelldict)
pvmod = PVmodule(pvcells=pvcell, **pvmoddict)
pvsys = PVsystem(pvmods=pvmod, numberStrs=num_strs, numberMods=num_mods, pvconst=pvmod.pvconst)
```

Does something like this sound like what you have in mind? I think we could make a method in PVsystem that creates a nested dictionary with this information and and option in `__init__()` that can reconstruct a system from the nested dictionary.

Thoughts?