SunPower/PVMismatch #52: use lookup table for 2-diode parameters (a la SAM)

mikofski, 2017-08-10:
related to #50 and #51 

At PVSC Janine Freeman introduced a great idea to nearly eliminate errors from diode-analog cell models, which was to use a lookup table.

Since we already have the `pvmismatch.contrib.gen_coeffs` methods, all we need is to loop over it so it creates an IEC-61853 table but for Isat1, Isat2, Rs and Rsh instead of Imp, Vmp, Isc, and Voc.

Then we would have to add the interpolation methods into `pvmismatch.pvcell` or at least the option to.

mikofski, 2017-08-10:
see [**Significant Improvement in Module Performance Prediction Accuracy Based on IEC-61853 Data** by Janine Freeman at PVPMC-8](https://www.slideshare.net/sandiaecis/06-20170509-freeman-8th-pvpmc-iec-61853-presentation)

mikofski, 2017-08-10:
see [Trello](https://trello.com/c/qpLucqHp)

xjianwei520, 2018-01-10:
Dear Mikofski : i saw the fuction about' gen_coeffs.gen_two_diode()',  it can return the cell parameter such as isat1, isat2, rs, rsh. actually, Some R&D Projects also need the cell reversed parameter such as ARBD,BRBD,VRBD,NRBD.  The current packages can solve this problem ?

mikofski, 2018-01-10:
@xjianwei520, no, I didn't add methods to fit reverse breakdown parameters, because AFAIK there isn't an IEC standard to test or measure for this, and often there is little or no published data on reverse bias characteristics of cells or modules.

But I agree that we should at least provide some guidelines or rules of thumb, maybe in the documentation or the wiki. Probably we should get a few other researchers to give their input?

The equations for the reverse breakdown are in issue #25 and in [`pvcells.py`](../blob/master/pvmismatch/pvmismatch_lib/pvcell.py#L169-L174).

Here's what I would suggest:

1. Try to find out what the breakdown voltage is and use that to set `VRBD`. Typically front contact p-type c-Si cells breakdown in reverse bias somewhere between -17.0-volts to -25.0-volts. SunPower modules are back contact n-type and they breakdown between -3.5 and -5.5 volts.

2. Try to get real reverse bias test data and fit it using [`scipy.optimize.curve_fit`](https://docs.scipy.org/doc/scipy/reference/generated/scipy.optimize.curve_fit.html) to get ARBD, BRBD and NRBD using the equation from #25 and . Otherwise, just use the defaults for for ARBD, BRBD, and NRBD - unfortunately, this model is not great, but there is an issue #26 to make a more flexible reverse bias model

xjianwei520, 2018-01-11:
Thanks ,Mark.
 I will try it follow your idea, and i also give you all feedbacks about result whatever good or bad. 
hoping the pvmismatch more and more powerful

thunderfish24, 2018-01-11:
What’s your fitting data look like? I have a “global” single diode model
fitting already in production, and I’m always curious to compare fit values
between single and double diode models.