SunPower/PVMismatch #57: band gap is not variable in contrib.gen_coeffs()

mikofski, 2018-01-10:
The band gap, `Eg`, is constant in [`diode.py` line 10](../blob/master/pvmismatch/contrib/gen_coeffs/diode.py#L10) but different tech would have a different band gap so it should be a variable.

thunderfish24, 2018-01-10:
I agree. I fit this parameter from a collection of IV curves in my parameter fitting software. Whether this fit values is physically meaningful vs. completely confounded is another another story ;).

mikofski, 2018-01-10:
Thanks @thunderfish24 would you be will to contribute something to the [`contrib.gen_coeffs` package](../tree/master/pvmismatch/contrib/gen_coeffs) ?

thunderfish24, 2018-01-11:
If I could get compensated for the initial effort to make my diode fitting
public domain, then I would strongly consider it. 🙂