SunPower/PVMismatch #66: PVModule IV curve has no knees

sdavila604, 2018-02-22:
Hi!

I am simulating shading on a STD72 module to test microinverter MPPT tracking.  I am passing the PVmodule.setSuns function a np.ndarray with values between 0.2 and 1.0 as a shade passes over the module.  However, the IV curve produced does not seem to have the "knees" that we would expect.

I am attaching an mp4 file of the animation, as well as the Python script I am using.  I apologize for how poorly structured and commented it is.  It is a work in progress (scaffolding!).

Any help or insight would be greatly appreciated!

[panel_animation.zip](https://github.com/SunPower/PVMismatch/files/1749684/panel_animation.zip)

[dynamic_shading.zip](https://github.com/SunPower/PVMismatch/files/1749687/dynamic_shading.zip)

chetan201, 2018-02-22:
@sdavila604 
I am not sure I follow you about the "knee" of the IV curve. I do see it in the animation you sent. Limiting the axes to [0,Isc] and [0,Voc] would show it better.

sdavila604, 2018-02-23:
Thank you for your very prompt response!
What we see on real panels when we shade sections of them is several bends (which we call "knees").  These knees result from the bypass diodes bypassing sections of the panel that are no longer generating current.  The microinverter MPPT tracking algorithm must hunt for the best "knee" to maximize power extraction.  Here is an image that shows what the IV curves look like with partial shading of panels:

[image]

mikofski, 2018-02-23:
you mean like this:
[image]
```python
>>> from pvmismatch import *
>>> pvm = pvmodule.PVmodule()
>>> pvm.setSuns([0.2, 0.2, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0,
... 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 0.2, 0.2, 1.0,
... 1.0, 1.0, 0.2, 0.2, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0,
... 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 0.2, 0.2, 1.0, 1.0, 1.0,
... 1.0, 1.0, 1.0, 1.0, 0.2, 0.2, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0,
... 1.0, 1.0, 1.0, 1.0, 1.0, 0.2, 0.2, 1.0, 1.0, 1.0, 1.0, 1.0,
... 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0,
... 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0])
>>> x = pvm.plotMod()
>>> x.show()
```

mikofski, 2018-02-23:
or this:
[image]
```python
pvm.setSuns([0.2, 0.2, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0,
1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 0.2, 0.2, 1.0,
1.0, 1.0, 0.4, 0.4, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0,
1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 0.4, 0.4, 1.0, 1.0, 1.0,
1.0, 1.0, 1.0, 1.0, 0.6, 0.6, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0,
1.0, 1.0, 1.0, 1.0, 1.0, 0.6, 0.6, 1.0, 1.0, 1.0, 1.0, 1.0,
1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0,
1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0])
y = pvm.plotMod()
y.show()
```

mikofski, 2018-02-23:
@sdavila604 I think what your image is showing is that there is:
* a substring that has almost zero shade, and hence can carry nearly 6 amps
* an because it carries that current nearly up to 30 volts or half the panel, then it **must be the center substring**
* a side substring that has 75% shade on enough cells that if they were in reverse bias, they would trigger a bypass diode, typically 2 to 3 cells (-5.5V*3cells = 16.5V > 24cells*0.55V = 13.2V)
* another side substring that has like maybe 20% shade on enough cells etc. (same as other side substring)

That's why you see that shade pattern. So reproduce this in pvmismatch:
```python
>>> from pvmismatch import *
>>> pvm = pvmodule.PVmodule()   # initial module has 1.0 suns everywhere
>>> pvm.setSuns([0.75, 0.75, 0.75, 0.25, 0.25, 0.25], [0, 1, 2, 92, 93, 94])
>>> x = pvm.plotMod()
>>> x.show()
```
and you get your knees:
[image]

sdavila604, 2018-02-23:
Thank you!

I ran your code and, sure enough, I get the same curve you show.

Just to make sure I am interpreting this correctly, you have cells 0,1,2 getting 0.75 suns and cells 92,93,94 getting 0.25 suns.  Correct?  I will see if I can get the same using my code and let you know.

Thanks!

sdavila604, 2018-02-23:
So, this is the closest I can get to duplicating your "setSuns" with my code:

Cell idx: 0 Suns: 0.752121440271867
Cell idx: 1 Suns: 0.7480000000000002
Cell idx: 2 Suns: 0.762620925038382
Cell idx: 3 Suns: 1.0
Cell idx: 4 Suns: 1.0
...
Cell idx: 90 Suns: 1.0
Cell idx: 91 Suns: 1.0
Cell idx: 92 Suns: 0.2462432599770664
Cell idx: 93 Suns: 0.2462432599770663
Cell idx: 94 Suns: 0.24624325997706706
Cell idx: 95 Suns: 1.0

Here is the (single frame) animation I get:
[panel_animation.zip](https://github.com/SunPower/PVMismatch/files/1753009/panel_animation.zip)

sdavila604, 2018-02-23:
The problem was my plot!  For some reason in pyplot if you only update the y coordinates it gets confused.

Thank you for your help and sorry for the distraction.

[panel_animation.zip](https://github.com/SunPower/PVMismatch/files/1753316/panel_animation.zip)

mikofski, 2018-02-24:
🎉 @sdavila604 does this resolve your conflict? Will you close this issue? Thanks

sdavila604, 2018-02-26:
It does.  Thank you.

closed 2018-02-26 by sdavila604