SunPower/PVMismatch #71: Enable bypass diodes in pvmodule class to be non-ideal diodes

chetan201, 2018-03-07:
Enable non ideality to bypass diodes in the pvmodule class

mikofski, 2018-03-12:
You can use [Shockley's equation](https://en.wikipedia.org/wiki/Shockley_diode_equation) for this. The values of the bypass diodes that were in use at SunPower from 2010 to 2015 or so are in [PVLife](https://github.com/SunPower/PVLife). If you PM me I can give you more info. Basically all you need is the saturation current, `I_sat` so `Ibypass = I_sat * (exp(q*Vbypass / n / kB / T_K) - 1.0).

But are you sure this is necessary? IMO the ideal diode behavior is probably accurate enough for most circumstances.

See also:
* [diode](https://en.wikipedia.org/wiki/Diode#Shockley_diode_equation)
* [diode modeling](https://en.wikipedia.org/wiki/Diode_modelling#Shockley_diode_model)

xjianwei520, 2018-04-02:
Dear mikofski :
I see the updated version about pvmismatch .  where  i can find  all updated information about version v4.0

mikofski, 2018-04-03:
Hi @xjianwei520 you should probably ask @chetan201 he is the current owner of PVMismatch. I don't work at @sunpower anymore.

@chetan201 do you plan on updating the docs and release a new version with the latest changes from @rayhickey ?

Also, maybe there should be a google group or someplace on the wiki that's more appropriate for user comments or questions such as this?

mikofski, 2018-04-05:
You're probably using the _module_ `Rsh` which is typically in the 100's. Whereas PVMismatch uses the _cell_ `Rsh` which is typically around 10&Ohms;

chetan201, 2018-04-06:
Hi @xjianwei520, we agree regarding @mikofski 's feedback. (Thanks @mikofski !) 

As for the release 4.0, I haven't had a chance to update the documentation yet but the key difference is to the Temperature-Voltage characteristics. We implemented temperature dependance to Isat2 value in the 2 diode equivalent model at the heart of PVMismatch. See below - 
[image]

As for your second question, let me take a quick crack here -  

1-gen_two_diode ( Isc , voc, Imp, vmp ,nseries, nparrllel, tc, x0=None, *args, **kwargs)
2-residule_two_diode(x , isc ,voc, imp ,vmp ,tc)

Function 1 and 2, essentially solves the two diode model equations for Rs, Rsh, Isat1, Isat2 values when given with the solutions to the two diode model equations (i.e. Isc, Imp, Voc, Vmp - the data sheet parameters)