SunPower/PVMismatch #74: improve irradiance response

mikofski, 2018-05-04:
# Problem:

When fitting parameters there seems to be a systematic bias with irradiance, typically increasing. This may be caused by PVMismatch's constant shunt resistance. Other models such as SAM/DeSoto 5-parameter or PVSyst have a shunt resistance that is a function of irradiance.

## Shunt resistance
[PVPVMC on De Soto “Five-Parameter” Module Model](https://pvpmc.sandia.gov/modeling-steps/2-dc-module-iv/diode-equivalent-circuit-models/de-soto-5-parameter-model/)

    Rsh = Rsh_0 * E0 / Ee

_Note_: irradiance is in the denominator so as irradiance increases the shunt resistance decreases.

# Proposal:
add a factor to account for irradiance bias base on DeSoto 5-parameter model

thunderfish24, 2018-09-28:
@mikofski Have you ever seen evidence that this change in shunt resistance is actually an irradiance phenomenon rather than a temperature phenomenon?

I have been fitting many curves in the mPERT outdoor datasets, and it's hard to say what the root cause of the change in shunt resistance is, because the temperature and irradiance are strongly correlated positively in these datasets.

Perhaps there are some IEC 61853-1 matrix measurements that clearly show the trend in the irradiance dimension but not in the temperature direction?

mikofski, 2018-09-29:
Both the PVSyst and DeSoto/CEC models have an irradiance dependence for shunt resistance although I don't what the physical explanation is. Kyumin didn't discuss it at all in his presentations, only the reporter dependence on series resistance do to metallization. I agree there is a strong correlation between