SunPower/PVMismatch #76: PR #73 breaks temperature dependance

chetan201, 2018-05-08:
What's expected - (reverting back to pr #68 
[image]

What's observed in pr #73  -
[image]

chetan201, 2018-05-08:
@mikofski  
off the top of your head, do you know why this might be happening? 
Shall I revert the master to PR #68 meanwhile?

mikofski, 2018-05-09:
Another fix idea, would be to just pick a 4th quadrant current, like -10A, and just go up to that value always. I'll try to implement one of these two approaches and put in a new PR asap. Until then, yes, I think you would be better reverting to 4943742f68a1819d258bf930d89c4b05f06b7630 

sorry!

commit 7229512e (2018-05-09): fixes #76 issue with very low temperatures (#77)

* if Voc at Tcell is actually greater than Voc at STC, then set Vmax to
Voc, and set 3rd set of points log spaced with decreasing density from
Voc at STC to the greater Vmax
* update comment
closed 2018-05-09 by chetan201