SunPower/PVMismatch #84: PVMismatch only using default 96 cell config

tamirlance, 2018-07-23:
Does anyone else have a problem with PVMismatch overriding a pvmodule object witht he default STD96 one (using PVMismatch 4.0)?  

The code below results in power for a 96 cell and not 128 cell module.  pvm is correct but when it gets to pvs the pvm gets overwritten with STD96 module.

from pvmismatch import PVconstants
from pvmismatch import *

def get_pvs(irr,temp,tech='E'):
    pvconstants = PVconstants(npts=10001)
    modules = {'E':{'Rs':0.0047299749332989785,'Rsh':11.288205142648826,'Isat1':1.9007562365404928e-11,'Isat2':1.044218374554042e-06,'Isc0_T0':6.549045},
               'X':{'Rs':0.0022904553199000655,'Rsh':5.524413919705285,'Isat1':2.6951679883577537e-12,'Isat2':9.078875806333005e-07,'Isc0_T0':6.590375}}

    pvc = pvcell.PVcell(Rs=modules[tech]['Rs'],Rsh=modules[tech]['Rsh'],Isat1_T0=modules[tech]['Isat1'],Isat2_T0=modules[tech]['Isat2'],
                            Isc0_T0=modules[tech]['Isc0_T0'],Ee=irr/1000.,Tcell=temp+273.15,pvconst=pvconstants)
    pvm = pvmodule.PVmodule(cell_pos=pvmodule.standard_cellpos_pat(16,[2,4,2]), pvcells=pvc, Vbypass=-0.5, pvconst=pvconstants)
    pvs = pvsystem.PVsystem(pvconst=pvconstants, pvmods=pvm, numberStrs=1, numberMods=1)
    return pvs

if __name__ == "__main__":
    pvs = get_pvs(irr = 1000., temp = 85., tech='E')

    pvs.setSuns(1)
    Pmp_0 = pvs.Pmp
    Vmp_0 = pvs.Vmp
    Imp_0 = pvs.Imp
    print Pmp_0, Vmp_0, Imp_0

chetan201, 2018-07-23:
Tamir,
Change 
pvs = pvsystem.PVsystem(pvconst=pvconstants, pvmods=pvm, numberStrs=1, numberMods=1)
to 
pvs = pvsystem.PVsystem(pvconst=pvconstants, pvmods=**_[pvm]_**, numberStrs=1, numberMods=1)

To get around this, I always use pass lists of objects rather than relying on PVMismatch using one template object for all. This could be improved though. I will make this an issue to be handled in the future.

Thanks!

mikofski, 2018-07-23:
Hi @tamirlance and @chetan201 , this bad behavior was because of a typo bug in `pvstring.py`  which was fixed in #83 sorry! If I use the latest master, your code works.

```python
(py36) PVMismatch>ipython
Python 3.6.6 |Anaconda, Inc.| (default, Jun 28 2018, 11:27:44) [MSC v.1900 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 6.4.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from pvmismatch import PVconstants

In [2]: from pvmismatch import *

In [3]: %paste
def get_pvs(irr,temp,tech='E'):
    pvconstants = PVconstants(npts=10001)
    modules = {
        'E': {
            'Rs': 0.0047299749332989785,
            'Rsh': 11.288205142648826,
            'Isat1': 1.9007562365404928e-11,
            'Isat2': 1.044218374554042e-06,
            'Isc0_T0': 6.549045
            },
        'X': {
            'Rs': 0.0022904553199000655,
            'Rsh': 5.524413919705285,
            'Isat1': 2.6951679883577537e-12,
            'Isat2': 9.078875806333005e-07,
            'Isc0_T0': 6.590375
        }
    }
    pvc = pvcell.PVcell(
        Rs=modules[tech]['Rs'],
        Rsh=modules[tech]['Rsh'],
        Isat1_T0=modules[tech]['Isat1'],
        Isat2_T0=modules[tech]['Isat2'],
        Isc0_T0=modules[tech]['Isc0_T0'],
        Ee=irr/1000.,
        Tcell=temp+273.15,
        pvconst=pvconstants
    )
    pvm = pvmodule.PVmodule(
        cell_pos=pvmodule.standard_cellpos_pat(16,[2,4,2]), pvcells=pvc, Vbypass=-0.5, 
        pvconst=pvconstants
    )
    pvs = pvsystem.PVsystem(pvconst=pvconstants, pvmods=pvm, numberStrs=1, numberMods=1)
    return pvs

## -- End pasted text --

In [4]: pvs = get_pvs(irr=1000., temp=85., tech='E')

In [5]: pvs.setSuns(1)

In [6]: Pmp_0 = pvs.Pmp

In [7]: Vmp_0 = pvs.Vmp

In [8]: Imp_0 = pvs.Imp

In [9]: print(Pmp_0, Vmp_0, Imp_0)
359.2639068472598 59.234953061689176 6.065066118531585
```

mikofski, 2018-07-23:
@chetan201 and @tamirlance , I'm pretty sure you can close this, since #83 fixed it

mikofski, 2018-07-31:
@tamirlance and @chetan201 - do you guys agree that the issue with making systems given a `PVmodule` was caused by the bug from #82, that #83 fixes? Do you think this issue still needs to remain open?

tamirlance, 2018-07-31:
Is there an official release with this fix that I can test?  I’ll take your word for it that #83 fixes this.  But I also found a new maybe unrelated bug

I have this new line in my code:

pvs = pvsystem.PVsystem(pvconst=pvconstants, pvmods=(pvm,)*nmods, numberStrs=1, numberMods=nmods) #Create a string with the number of modules specified

where nmods is a scalar for number of modules in the string.  It seems like numberMods and numberStrs are superfluous but it does not work without them.  It defaults to 10 modules per string without them

I’m not sure what happens if there is a disconnect between pvmods and numberStrs/numberMods but I think as long as numberStrs/numberMods isn’t blank then they will get overwritten by pvmods (but I haven’t checked this yet)

Tamir

mikofski, 2018-08-02:
@tamirlance you can use [pip in any Python to do an install straight from GitHub](https://pip.pypa.io/en/stable/reference/pip_install/#git):

    $ pip install git+https://github.com/SunPower/PVMismatch#egg=PVMismatch

[The new bug in your comment](#issuecomment-409315552) refers to the exact same bug as what is described in #82 and fixed in #83. Basically what is happening is that since the attempt to get `pvconst` from the module is inside a try-except block, and it fails in your version because there is a typo, then it falls back on the default `PVconstants()` instance which always has ten `STD96` module.

See [the code changed in PR83](../../../blob/46b65ee15bec151b78b0c7ad375167b4dc468fa3/pvmismatch/pvmismatch_lib/pvstring.py#L35-L42)
```python
            # is pvmods an object?
            try:
                pvconst = pvmods.pvconst
            except AttributeError:
                #  try to use the pvconst arg or create one if none
                if not pvconst:
                    pvconst = PVconstants()
                # create pvmod
                pvmods = PVmodule(pvconst=pvconst)
            # expand pvmods to list
            pvmods = [pvmods] * numberMods
```

If I use the master branch it works fine for me:

```python
>>> from pvmismatch import *
>>> pvm = pvmodule.PVmodule()
>>> nmods = 18
>>> pvconst = pvconstants.PVconstants()  # do NOT use pvconstants b/c that is a module!
>>> pvs = pvsystem.PVsystem(pvconst=pvconst, pvmods=(pvm,)*nmods, numberStrs=1, 
...     numberMods=nmods) #Create a string with the number of modules specified
>>> len(pvs.pvstrs[0].pvmods)
18
```

chetan201, 2018-08-14:
@mikofski 
thanks for clarifying this. 

A release is certainly overdue for PVMismatch. I will try to shoot for an early Sep release with the bypass diodes configurability.

closed 2018-08-14 by chetan201