SunPower/PVMismatch #88: Make PVMismatch work with all the modules in the Sandia/CEC database

chetan201, 2018-09-17:
This is a broad direction we'd like to take PVMismatch to. As far as I know, this hasn't been tried. 

The goal would be to make PVMismatch being able to generate PVcell objects from CEC/Sandia database used in PV modeling community. 

The ```gen_coeff``` method under contrib needs to be tested for goodness of fit for the entire database of the modules.

thunderfish24, 2018-09-18:
I have a plan to implement model comparison metrics with automatic model selection at pv-fit.com. I would also like to offer some sort of model discrepancy metric to alert users that the model might be significantly wrong. There are other issues, such as normalizing the metrics to the current level and noise level in the data. I would very much like to help establish some commonly accepted metrics for these use cases, and I could suggest some to start with for diode models.

mikofski, 2018-09-18:
IMHO you should focus on IEC81653 as the standard set of conditions, and the module characterization, then all model specific characterizations derive from that

thunderfish24, 2018-09-19:
@chetan201 There is also http://pvmodel.umh.es

Also, would you say that you are looking to somehow validate models under mismatched conditions too? (Pardon my ignorance of what this actually entails for PVMismatch.)

thunderfish24, 2018-09-23:
One measure of goodness-of-fit (or lack of model discrepancy) that I use for diode models is the residuals of the sum of currents at the high-voltage diode node in the equivalent circuit model. Via Kirchhoff's Current Law and the DC model assumptions, this sum should be zero for every noise-free I-V data point. Here is a figure for a single-diode model (SDM) fit and a double-diode model (DDM) fit to a single I-V curve at one temperature and irradiance (what I call a "local" model fit). I often see the oscillatory residuals in the SDM fits, and such model discrepancy only gets worse when fitting an IEC-81653 "matrix" of curves to a "global" SDM or DDM model [[reference](https://onlinelibrary.wiley.com/doi/full/10.1002/ese3.190)].

[image]

thunderfish24, 2018-09-24:
@mikofski Does @cwhanse follow this repo?

Cliff: If you see this, then I think it might be good to dust of that model fitting inter-comparison study you did a while back. I think two general categories of test problems would be useful: (1) Fitting to single I-V curves from the IEC matrix of I-V curves with the "standard" SDM or DDM equation and noise in I and V channels, and (2) Fitting to the whole set of IEC matrix of I-V curves, and this could include only using the "sparse" data, i.e., just the Isc, Pmp, and Voc points.

I think (2) actually would have a lot of technical issues to sort through, but (1) should be doable ATM. I think (1) should include fitting the SDM to data generated by the DDM, so as to inspect the effect that model discrepancy has on one's SDM fitting algorithm, and we could try to set up a similar model discrepancy test for the DDM fits. As discussed above, we would want to establish the goodness-of-fit metrics up front.

cwhanse, 2018-09-24:
I follow it now 😊

That inter-comparison wasn’t completed due to project funding being redirected.  I have thought for quite a while that, for algorithm validation, there is a need for benchmark problems with known solutions to test algorithms.  Something similar to https://sparse.tamu.edu/.

Knowing that an algorithm can recover known parameters is a prerequisite to judging that the algorithm’s skill at estimating unknown parameters.

chetan201, 2018-09-27:
> @chetan201 There is also http://pvmodel.umh.es
> 
> Also, would you say that you are looking to somehow validate models under mismatched conditions too? (Pardon my ignorance of what this actually entails for PVMismatch.)

@thunderfish24 
I meant from the perspective of being able to match IEC matrix results as @mikofski  pointed out. I think if there is good fit, that is a great step in itself.

thunderfish24, 2018-09-27:
@chetan201 I am working on an update to PVfit’s fitting algorithm because of issues my current solver stumbles on with the double-diode model. Based on the changes, I am ready to revisit fitting to a sparse IEC matrix, i.e., fitting a full temperature and irradiance dependent performance model having only the Isc, Imp, Vmp, and Voc points.

@cwhanse @mikofski Can you verify that the spectrum in the IEC matrix is the same one as used for STC? Likewise, what is the temperature of, the junction, average cell, back-of-module, or ambient?

cwhanse, 2018-09-27:
Depends on how the curves were measured.  From recollection, IEC61853 does not specify equipment or precise conditions beyond temperature and irradiance values.  It is still important to ask how the irradiance is measured, what controls or calibration are used for the lighting, how temperature is imposed and measured, etc.

thunderfish24, 2018-09-28:
@cwhanse Right. That's why above I said "I think (2) actually would have a lot of technical issues to sort through", where (2) refers to "Fitting to the whole set of IEC matrix of I-V curves, and this could include only using the "sparse" data, i.e., just the Isc, Pmp, and Voc points." For example, PVfit prefers to use "uncorrected" I-V-F-T curve data that are readily available from indoor calibration laboratory "matrix" measurements, but often not made available to the data "consumer". These corrections can introduce bias into the data, so my opinion is don't use them.

cwhanse, 2018-09-28:
Checked my copy of IEC61853-1, the intent is that the measurements reflect "AM1.5 global" spectrum from IEC60904-3.

IEC61853-1 requires class BBB or better simulator, and allows for outdoor IV measurements. It points to IEC60904-7 for adjusting for spectral mismatch between module and irradiance device, and presumably between irradiance and normative spectrum. But like many of these standards, there is room for interpretation.

markcampanelli, 2019-05-13:
@cwhanse @mikofski @chetan201 Is there any chance that we could discuss establishing "standard" goodness-of-fit metrics on Wednesday afternoon at the performance Workshop? I think that it might be possible to define inter-comparison metrics for diode-based DC models based on Kirchoff's current conservation law. Of course, there may be other priorities that I'm not aware of. It's sad that DoE apparently hasn't seen value in funding this effort to a successful completion in the past.

cwhanse, 2019-05-13:
I won't be at the workshop Wednesday. Let me know if you are able to meet.

chetan201, 2019-05-16:
@markcampanelli Sorry just seeing your comment here. But glad we had a chance to touch base briefly. I would like to continue the discussion and plan to look at the mPERT data towards goodness of fit.  (@mikofski 's suggestion ) 
@cwhanse sorry to hear that DOE hasn't furthered funding to this topic. In my opinion, the impact of gaining accuracy here is really big.