SunPower/PVMismatch #101: PVMismatch Capability for CellPos of Uneven parallel substrings

tamirlance, 2019-05-14:
Is it possible to have multiple substrings in parallel in a module where the substrings have different number of cells in series?  For example

Imagine a 59 cell module which is made of 6 strings in parallel.  5 strings have 10 cells in series and the 6th one has 9.

I think there should be a way to do this with crosstied cell position structures but not sure exactly how to execute

mikofski, 2019-05-17:
If the substrings were in parallel, then AFAIK this would cause harmful mismatch, so IMO I don't think a manufacturer would make a panel like this, unless those 9 cells were different (different voltage) than the other cells.

But if they were in series then that would be okay, and I think that the current PVMismatch bypass diode settings let you select the number of cells in each substring already - ok almost, not flexible enough for 59 cell module, because `pvmodules` assumes that cells form a full rectangular grid, and 59 cells would have one empty, so this would be a new feature I guess.

tamirlance, 2019-05-17:
Thanks Mark.  For sure this would not be for an intentional design.  It would help us model what happens when cells are shorted and not contributing to the circuit.  I guess this would be a new feature request - to handle non-rectangular configurations.  Right now we are trying to do a  workaround by modelling the shorted cell using the parameters available in PVMismatch (i.e. Isat1, Isat2, Rsh, Rs, etc).  But it would better if we could just omit the cell entirely as you get into weird regimes trying to shoehorn a cell model into mimicking what is essentially a simple conductor

mikofski, 2019-05-17:
>For sure this would not be for an intentional design. It would help us model what happens when cells are shorted and not contributing to the circuit. 

so this is already possible to model in pvmismatch as it is right now, you basically want to get no voltage response at any current, so you can carry the module's current with zero or no voltage drop:
1. create a standard 60 cell module
2. set that one cell irradiance close to zero (turns `Iph` to zero) and copies (memoizes) that cell
3. set that one cell `Isc0_T0` close to zero
4. set that one cell `IsatN_T0` (saturation/dark current) close to zero  (turns the diode current off)
5. either leave `Rs` and `Rsh` alone, and you'll get a voltage of `Vc = -Ic*(Rsh + Rs)` or set them to whatever you want - note that the voltage will always be negative because you are dissipating energy

For example:
```python
from pvmismatch import *
from matplotlib import pyplot as plt

# make a standard 60-cell module with 3 substrings, 20 cells ea
STD60 = pvmodule.standard_cellpos_pat(10, [2, 2, 2])
pvmod = pvmodule.PVmodule(cell_pos=STD60)
ncells = len(pvmod.pvcells)  # should be 60 cells
assert ncells == 60

# plot the normal module and cells
plt.ion()
pvmod.plotCell()
pvmod.plotMod()
```
[image]
[image]

```python
# use setSuns to make a copy of the last cell and set the irradiance to nil
pvmod.setSuns(cells=[59], Ee=0.001)

# now the last cell is a copy, we can change it's properties to short it
# set the short circuit current, saturation currents, series resistance to nil
# and set the shunt resistance to a bigger number and kill reverse breakdown
pvc_last = pvmod.pvcells[-1]
pvc_last.update(Isc0_T0=1e-16, Isat1_T0=1e-16, Isat2_T0=1e-16, Rs=1e-16, Rsh=1e16, aRBD=0, bRBD=0)
pvc_last
# <PVcell(Ee=0.001[suns], Tcell=298.15[K], Isc=1e-19[A], Voc=1.71236e-05[V])>
```
```python
# we need to recalculate the module again, easiest way is with setSuns:
pvmod.setSuns(cells=[59], Ee=1e-16)
pvmod.plotCell()
pvmod.plotMod()
```
[image]
[image]

... which is what I would expect, if you short the last cell, then you get a module that has a lower voltage and less power.

mikofski, 2019-05-17:
@chetan201

tamirlance, 2019-05-17:
Thanks Mark - this makes sense.  I wouldn't have expected that shorting one cell would have had such a dramatic impact on the power.  Shouldn't you just lose the power of the one cell - assuming you are in an all series circuit?  It looks like the diode is kicking in here which indicates reverse bias activation of the cell.

mikofski, 2019-05-18:
I saw that too,  you may need to fool around with it, looking at that one cell,  it looks like I set the resistance so high that it can't carry any current,  so perhaps try again with a more reasonable resistance and that should fix it. I spent too much time on this, but it was fun although perhaps you can take our from here now that you understand how to set the cell properties using `update` and to recalculate the module using `setSuns()`

FYI you can also call
```python
pvm.Imod, pvm.Vmod, pvm.Pmod, pvm.Isubstr, pvm.Vsubstr = pvm.calcMod()
```
Where `pvm` is the module

The documentation needs some work,  interested?

chetan201, 2019-05-20:
@mikofski 
Thank you so much for your guidance here!