SunPower/PVMismatch #107: PV system layout and cell irradinace input/output via xls

adambgnr, 2019-07-25:
It is fun to experiment with shading cells in PVMismatch and see, how different patterns affect the power output. On the other hand, it is not easy to set specific shade patterns on the system manually and be reassured that the system layout looks as intended, especially for new users. I was looking for a tool, that can do this and is: lightweight, accessible and easy to use. To me Excel or one of its free alternatives seems like a viable option. I have implemented a prototype of this functionality here: 
[https://gitlab.tue.nl/bp-tue/pvmismatchxlsio](https://gitlab.tue.nl/bp-tue/pvmismatchxlsio)
Would exporting the PV system layout, and importing cell irradiance from formatted xlsx files be a useful addition to PVMismatch?

mikofski, 2019-08-08:
Hi @adambgnr, perhaps consider trying the [module mismatch simulator](../blob/master/pvmismatch/contrib/module_mismatch_simulator.py) in `contrib` developed by @bmeyers 

[image]

mikofski, 2019-08-08:
> I have implemented a prototype of this functionality here:
https://gitlab.tue.nl/bp-tue/pvmismatchxlsio

Looks great!

>Would exporting the PV system layout, and importing cell irradiance from formatted xlsx files be a useful addition to PVMismatch?

Yes I think so. I wonder what @chetan201 thinks? Would you consider adding this to `contrib` or just keep it separate but referenced in the PVMismatch documentation?

adambgnr, 2019-08-09:
> Would you consider adding this to contrib or just keep it separate but referenced in the PVMismatch documentation?

I would be happy to contribute it to `contrib`.

adambgnr, 2019-08-09:
> Hi @adambgnr, perhaps consider trying the [module mismatch simulator](../blob/master/pvmismatch/contrib/module_mismatch_simulator.py) in `contrib` developed by @bmeyers
> 
> [image]

This is very useful. I haven't stumbled upon this while looking trough the repository. I could look into adding the functionality of indicating activated bypass-diodes to the excel interface as well.

bmeyers, 2019-08-09:
@adambgnr Thanks for your interest! The module mismatch simulator does indicate bypass-diode activation as well as reverse biasing of cells (for technologies where that is possible). Bypass diode activation outlines the cells in blue and reverse biasing outlines the cells in red.

mikofski, 2019-08-10:
>I would be happy to contribute it to contrib.

Excellent! Can you open a pull request?

adambgnr, 2019-08-12:
> Excellent! Can you open a pull request?

I have made the pull request. Let me know if you would like to change something in the TODO list.

chetan201, 2019-11-06:
PR merged

closed 2019-11-06 by chetan201

AugustVan, 2024-04-03:
@adambgnr is it also possible to use the code u made with a half cell panel like this?
[image]