SunPower/PVMismatch #110: "Resolution" of the I-V plots made with plotSys() changes with cell temperature

adambgnr, 2019-09-10:
If I set the cell temperatures high, then the "resolution" of the I-V and Power plots made with plotSys() decreases. It can be quite extreme if there are many different cell temperatures in the model:
[image]

mikofski, 2019-09-10:
Hi @adambgnr 

I have reproduced this issue, and I think I've traced the cause, but I'm still working on it.

In the meantime, there is a workaround. You can increase the resolution by creating a custom `PVconstants` instance with `npts=1000`

```python
from pvmismatch import *
pvconst = pvconstants.PVconstants(npts=1000)
pvsys = pvsystem.PVsystem(pvconst=pvconst)
pvsys.setTemps(388)
f = pvsys.plotSys();
f.show()
```
[image]

mikofski, 2019-09-10:
I traced the issue to [`pvconstants.calcSeries()`](../blob/master/pvmismatch/pvmismatch_lib/pvconstants.py#L121) specifically the lines that set the currents to interpolate voltage at - the minimum current in this example is HUGE and I think it skews the interpolation.

https://github.com/SunPower/PVMismatch/blob/faca92f7d6036aca2564907224a8de5e3ad9e38a/pvmismatch/pvmismatch_lib/pvconstants.py#L139-L144

The resolution at the cell level is actually fine!
[image]

But the resolution problem starts at the module level, so that's what got me digging into [`pvmodule.PVmodule.CalcMod`](../blob/master/pvmismatch/pvmismatch_lib/pvmodule.py#L426) which calls `pvconstants.calcSeries`.

In the cell, the resolution is set in [`pvcell.PVcell.calcCell`](/blob/master/pvmismatch/pvmismatch_lib/pvcell.py#L209) and I think the trouble starts with `Vquad4` which is a range of diode voltages that try to ensure that there is sufficient coverage to interpolate cells of different temperatures:

https://github.com/SunPower/PVMismatch/blob/faca92f7d6036aca2564907224a8de5e3ad9e38a/pvmismatch/pvmismatch_lib/pvcell.py#L214-L230

The issue, I believe is that although the diode voltage is set to Voc at STC, this results in a cell voltage that is much larger! EG:

```python
>>> pvsys_pvstr0 = pvsys.pvstrs[0]  # first string (str-0)
>>> pvsys_pvstr0_pvmod0 = pvsys_pvstr0.pvmods[0]  # first module (mod-0. str-0)
>>> pvsys_pvstr0_pvmod0.pvcells[0].VocSTC  # Voc@STC of first cell in (mod-0, str-0)
0.6744401939866893  # upper limit used in Vquad4

>>> pvsys_pvstr0_pvmod0.pvcells[0].Vcell[-1]  # last point in I-V curve for (cell-0, mod-0, str-0)
array([3.18693553])

>>> pvsys_pvstr0_pvmod0.pvcells[0].Icell[-1]  # last point in I-V curve for (cell-0, mod-0, str-0)
array([-588.78742012])

>>> pvsys_pvstr0_pvmod0.pvcells[0].Rs  # series resistance
0.004267236774264931

>>> pvsys_pvstr0_pvmod0.pvcells[0].Rs * pvsys_pvstr0_pvmod0.pvcells[0].Icell[-1]
... + pvsys_pvstr0_pvmod0.pvcells[0].Vcell[-1]
array([0.67444019])  # Vdiode = Vcell + Icell * Rs
```
So you can see that the setting the diode voltage to the estimated Voc at STC does accomplish the goal of being well into the 4th quadrant, it's TOO FAR! And that results in a minimum current, `Imin` in the `pvconstants.calcSeries()`, that is TOO BIG!.

mikofski, 2019-09-10:
I can confirm that Imin is the culprit. If I simple set Imin to zero, then the issue is resolved for the specific example I used in this issue, but I wonder if there are any edge cases I need to consider?
[image]

OK, I have talked to the master (@jdnewmil) and he reminded me of 2 things:
1. I am frequently asked what if there is negative current in the string, so the answer to my question about edge cases is yes, and I propose we do the same thing we do for voltage in the cell, and divide the current into 3 regions: reverse, forward, and negative (or quad4)
2. We better check that `pvconstants.calcParallel()` also works correctly too, because if this problem exists for series cells, then it might for cross-ties as well.

mikofski, 2019-09-11:
I have a PR ready, but didn't have a chance to push it yet, in case anyone is looking at this, I'll push it in a few hours

PR #112 (2019-10-31): BUG: Extend current in calcSeries to quad4
- fixes #110 - bad resolution for IV curve at high temperature
- change Iforward range to be from 0 to meanIsc, instead of from Imin
which was TOO SMALL, and use pvconst.Imod_pts which has closer spacing
near zero than pvconst.pts
- add quad4 range from Imin to zero, where Imin is a negative current in
the 4th quadrant, using pvconst.Imod_negpts which goes from 1 to 0,
with closer spacing near zero
- concatenate Iquad4 before Iforward, and increase Vtot to same size
- also add new pvconst.Imod_pts_sq which has even closer spacing near
zero to use with [... cut]
commit 5c4bc75f (2019-10-31): BUG: Extend current in calcSeries to quad4 (#112)

* WIP/BUG: Extend current in calcSeries to quad4

- fixes #110 - bad resolution for IV curve at high temperature
- change Iforward range to be from 0 to meanIsc, instead of from Imin
which was TOO SMALL, and use pvconst.Imod_pts which has closer spacing
near zero than pvconst.pts
- add quad4 range from Imin to zero, where Imin is a negative current in
the 4th quadrant, using pvconst.Imod_negpts which goes from 1 to 0,
with closer spacing near zero
- concatenate Iquad4 before Iforward, and increase Vtot to same size
- also add new pvconst.Imod_ [... cut]
closed 2019-10-31 by chetan201