SunPower/PVMismatch #111: error while using function call for PV MISMATCH

mdwasimatiq, 2019-09-10:
I Have bifacial radiance data in form of excel sheet generated from bifacial radiance software. 
I tried to import this data into PVMismatch using function call to find mismatch losses.
Unfortunately...it is showing following error in ipython console:

**ValueError: cannot reshape array of size 0 into shape (0,newaxis)**

Does anyone knows what this error means....and how it can be rectified??

**The complete Text of ipython console is given below:**

Python 3.7.3 (default, Mar 27 2019, 17:13:21) [MSC v.1915 64 bit (AMD64)]
Type "copyright", "credits" or "license" for more information.

IPython 7.4.0 -- An enhanced Interactive Python.

runfile('C:/Users/Wasim Atiq/Documents/bifacial_radiance-master/bifacial_radiance/function call.py', wdir='C:/Users/Wasim Atiq/Documents/bifacial_radiance-master/bifacial_radiance')
1 files in the directory
Traceback (most recent call last):

  File "<ipython-input-1-eee550dca3e0>", line 1, in <module>
    runfile('C:/Users/Wasim Atiq/Documents/bifacial_radiance-master/bifacial_radiance/function call.py', wdir='C:/Users/Wasim Atiq/Documents/bifacial_radiance-master/bifacial_radiance')

  File "C:\Users\Wasim Atiq\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 786, in runfile
    execfile(filename, namespace)

  File "C:\Users\Wasim Atiq\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/Wasim Atiq/Documents/bifacial_radiance-master/bifacial_radiance/function call.py", line 17, in <module>
    analysis.analysisIrradianceandPowerMismatch2(testfolder, writefiletitle, numpanels, sensorsy, portraitorlandscape)

  File "C:\Users\Wasim Atiq\Documents\bifacial_radiance-master\bifacial_radiance\analysis.py", line 368, in analysisIrradianceandPowerMismatch2
    [frontres, backres] = load.deepcleanResult(data, sensorsy, numpanels, automatic=True)

  File "C:\Users\Wasim Atiq\Documents\bifacial_radiance-master\bifacial_radiance\load.py", line 357, in deepcleanResult
    f_linear = interp1d(x_0, panelB['Wm2Front'])

  File "C:\Users\Wasim Atiq\Anaconda3\lib\site-packages\scipy\interpolate\interpolate.py", line 470, in __init__
    self._y = self._reshape_yi(self.y)

  File "C:\Users\Wasim Atiq\Anaconda3\lib\site-packages\scipy\interpolate\polyint.py", line 111, in _reshape_yi
    return yi.reshape((yi.shape[0], -1))

ValueError: cannot reshape array of size 0 into shape (0,newaxis)

In [2]:   

**The program code which i written for function call is**:

import bifacial_radiance
#cd C:\Users\Wasim Atiq\Documents\bifacial_radiance\bifacial_radiance
import analysis # might need to be on the folder to do this since it's not linked in the module
 
testfolder = r'C:\Users\Wasim Atiq\Documents\Test Ethiopia 1\results'
writefiletitle = r'C:\Users\Wasim Atiq\Documents\PV mismatch\Ethiopia\Gad1.csv'
numpanels = 2  # What you originally had on your simulation
sensorsy = 9   # ideally number close to what you originally had on your simulation
portraitorlandscape = 'portrait'
analysis.analysisIrradianceandPowerMismatch2(testfolder, writefiletitle, numpanels, sensorsy, portraitorlandscape)

Please help!!

mikofski, 2019-09-10:
Hi @mdwasimatiq

Thanks for posting your issue, and for providing the code and error traceback. From what I see, the issue is occuring in [bifacial-radiance](../../../NREL/bifacial_radiance) in [`load.py` on line 357 in `deepcleanResults`](../../../NREL/bifacial_radiance/blob/master/bifacial_radiance/load.py#L357):
```
File "C:\Users\Wasim Atiq\Documents\bifacial_radiance-master\bifacial_radiance\load.py", line 357, in deepcleanResult
f_linear = interp1d(x_0, panelB['Wm2Front'])
```

That error:
>ValueError: cannot reshape array of size 0 into shape (0,newaxis)

It usually means a broadcasting error - or NumPy and SciPy were expecting a different size array. In your case, the size is zero, or your array has nothing in it. So either the value of `x_0` or `panelB['Wm2Front']` is empty. I would [put a breakpoint in your code](https://docs.python.org/3/library/pdb.html), or [use `%pdb` in IPython](https://ipython.org/ipython-doc/3/interactive/reference.html#using-the-python-debugger-pdb) to have it stop at the trouble spot, and then figure out why those arrays are empty if you expected them to contain data.

Perhaps @shirubana might have some insight?

Good luck!

mdwasimatiq, 2019-09-22:
Thank you @mikofski for your explanation...
Actually iam new to all this and I dont have any idea how to use %pdb. I would like to know why it is necessary to put a breakpoint in my code.

Also i want to know "value of x_0 or panelB['Wm2Front']" is referring to what in this case? Is this referring to the column 'Wm2Front' in my testfolder? If its so, then its not empty...there are nine values below it.
Pls let me know if iam wrong.

Thanks in advance.

mikofski, 2019-09-22:
Hi @mdwasimatiq,

Sorry, in a nutshell, **you've posted your question in the _wrong_ repository**. Your heading says, "error while using function call for PV MISMATCH" but there is no call to PVMismatch in your code. Instead I see that you are importing and calling `bifacial_radiance` and that the exception is being raised by [`bifacial_radiance` in `load.py` on line 357 in the function `deepcleanResult`](../../../NREL/bifacial_radiance/blob/master/bifacial_radiance/load.py#L357), so **my advice would be to _start there_, in [bifacial-radiance](../../../NREL/bifacial_radiance)**. A look at your stacktrace shows the caller just before this exception is in [`analysis.py` line 368 in the function `analysisIrradianceandPowerMismatch2`](../../../NREL/bifacial_radiance/blob/master/bifacial_radiance/analysis.py#L17).

    File "C:\Users\Wasim Atiq\Documents\bifacial_radiance-master\bifacial_radiance\analysis.py", line 368, in analysisIrradianceandPowerMismatch2
    [frontres, backres] = load.deepcleanResult(data, sensorsy, numpanels, automatic=True)

Seaching for the argument `testfolder` in `analysis.py` it's used on [line 137 in a call to `read1result`](../../../NREL/bifacial_radiance/blob/master/bifacial_radiance/analysis.py#L137) which is called for each file in the test folder. **Perhaps put a breakpoint there and step through?**

>Also I want to know what "value of `x_0` or `panelB['Wm2Front']`" is referring to what in this case? Is this referring to the column 'Wm2Front' in my testfolder?

Sorry, you will have to ask @cdeline or @shirubana, since that's part of [bifacial-radiance](../../../NREL/bifacial_radiance), not PVMismatch. **Best to post your questions and issues there.**

>Actually I am new to all this and I don't have any idea how to use `%pdb`.

Welcome to computing. Since you're using Spyder, not IPython directly, you may not need to use `%pdb` which turns on interactive debugging in IPython and let's you step through your code line by line to find errors and try to solve problems. Instead you can use debugging interactively from the Spyder debugging menu. For more info read [the Spyder documentation on debugging](https://docs.spyder-ide.org/debugging.html#debugging-with-ipdb).

>I would like to know why it is necessary to put a breakpoint in my code.

Using breakpoints let's you stop your code and find out why it isn't working. For example, you could check the value of `panelB['Wm2Front']` and `x_0` while debugging.

Sorry I can't help more. Good luck!

shirubana, 2019-09-23:
@mdwasimatiq, @cdeline @mikofski 
I was sure I repied to this on 9/11, not sure why it didn't show up. Here is my answer again:

Hi,

We majorly updated the development branch last week and I think we were planning on doing a release that will improve the analysis.py functions for mismatch and the interaction with PVMismatch majorly. That being said, I don’t think the load function itself changed. It is hard without the files or knowing what is in the folder being loaded but here are some notes:

- 	Deepclean gets called inside of the analysisIrradiancePowerMismatch function on analysis.py. Right after read1result. That means that read1result is returning garbage because it’s attempting to read a file that is not in the right format :
- 	If there is **ANYTHING** in the testfolder passed besides the irr_**.csv result files from a bifacial radiance run, this function will not know and will try to do stuff and then deepclean will crash. (I’m attaching a good sample result for example)
- 	What this function does is it reads the result file, and cleans for materials like tube, sky, and edges of the panel that it encountes. This happens mostly when you are using a very high number of sampling, and when you have a y-gap between 2+ up modules.
- 	It seems that therefore either there is a file that is being tried to read by the function that does not have Wm2Front, or that Wm2Front is empty … it usually has values below.
- 	I would really, really suggest switching to using our development version, since the analysisIrradianceandPowerMismatch was heavily improved. Majorly. It’s been slow process but we have some great results from this improvements. We were hoping to release this before EUPVSEC or the BifiPV workshop but more probably by end of September

[image]

chetan201, 2019-11-02:
Seems like this question is resolved, closing the issue.

closed 2019-11-02 by chetan201