SunPower/PVMismatch #114: Files missing from contrib.gen_coeffs?

cwhanse, 2019-10-29:
`example.py` has references to `gen_coeffs.PVMODULES`, `gen_coeffs.gen_two_diode`, and other `gen_xxx` functions.  Am I missing something, or is PVMismatch missing something?

mikofski, 2019-10-30:
They're here https://github.com/SunPower/PVMismatch/blob/master/pvmismatch/contrib/gen_coeffs/__init__.py

cwhanse, 2019-10-31:
Thanks. Is it typical to put code in the __init__.py file?  I didn't think to look there.

mikofski, 2019-10-31:
Sorry, I'm not sure what's Pythonic. I think it's okay if they're constants and methods that apply to the entire sub-package, but I may be going rogue. I've learned a lot since PVMismatch was started like 7 years ago, so maybe I wouldn't do that again. It just irks me that the dunder module has to be there, but is just empty doing nothing, seems like a waste, so I usually try to use it somehow, but maybe that's wrong?

I definitely think it's okay to use `__init__.py` for creating or maintaining an API to sub-modules and packages beneath it the way pvlib does now, or to contain `__version__` and `__author__` information if it's at the top of the package, for use in `setup.py` and for users to extract for meta info.

cwhanse, 2019-10-31:
> I'm not sure what's Pythonic.

Neither am I.  "wrong" is a strong word.

Since you brought up python conventions, what is the appetite for migrating towards PEP8? e.g. function names from capital letters to lower case, etc.

mikofski, 2019-10-31:
Ugh, so many conflicts. I think this is a @chetan201 question, but IMO we should make this code [black](https://black.readthedocs.io/en/stable/), even if it has to go through deprecations or break code. Probably shouldn't be too painful since there are probably so few users.

cwhanse, 2019-10-31:
If it's OK to really depart from the current release, my project would support an overhaul of PVMismatch. Just tossing the idea out there. In the end our projects need a computational engine for mismatched IV curve calculations, that interfaces readily with pvlib, and which will persist for a long time (we'll re-use it).

mikofski, 2019-12-11:
Hi Cliff, maybe change the title of this issue to, "Move code from __init__.py, use better name" or perhaps consider closing it?

closed 2019-12-11 by cwhanse