SunPower/PVMismatch #122: CIGS cells under partial shading - difference with experimental results

a-karanth, 2020-04-27:
I am trying to simulate thin film CIGS cells in a 24x4 layout under partial shading. They are to be connected in series parallel (SP) as 4 strings of 24 cells and in total cross tied (TCT) configurations. For the shading pattern I am simulating, I also have experimental I-V curves for the same configurations. The cell in the code has been defined based on the cells used during experimentation. The shading pattern in this example is when cells in the bottom 12 rows are shaded, as shown in the image below. 
 
[image]

The simulation results from PVMismatch do not match that of the experimental results. I have attached plots of the SP and TCT I-V and power-voltage curves. The solid lines are the I-V curves and the dashed lines are the P-V curves.
   
[image]
[image]

As you can see, the position of the maximum power point matches close to the experimental results. However, later part of the curve is rather different from what is expected. I have also changed the diode to be modelled based on the schockey diode equation as was mentioned https://github.com/SunPower/PVMismatch/issues/71 but that did not affect the results much. For this shading pattern, when I look at the cells that are bypassed, that too matches with the expectations, ie, cells in the bottom 12 rows will be bypassed. Yet the I-V curves does not match.
Could you please help me with understanding why this issue might be occurring and how it can be overcome? It would be really help me move forward in my project.

I have also attached my code if you need it for reference.
[shade_query.txt](https://github.com/SunPower/PVMismatch/files/4541140/shade_query.txt)

mikofski, 2020-05-02:
First, thank you for trying PVMismatch! Second, I am sorry that the cell-pattern is so hopelessly confusing. It really needs a better API! See #113, #118, and #120 .

Third, I don't know if this matters, but I believe your series parallel string may not have the correct cell pattern. From the attachment, `shade_query.txt` there is:
```python
from pvmismatch import *
ncols_per_substr = [2]*12  # for SP
nrows = 1
#1 used for series parallel system
pv_mod_pattern1 = pvmodule.standard_cellpos_pat(ncols_per_substr=
                            ncols_per_substr, nrows=nrows)
```
this yields:
```
[[[{'crosstie': False, 'idx': 0}], [{'crosstie': False, 'idx': 1}]],
 [[{'crosstie': False, 'idx': 2}], [{'crosstie': False, 'idx': 3}]],
 [[{'crosstie': False, 'idx': 4}], [{'crosstie': False, 'idx': 5}]],
 [[{'crosstie': False, 'idx': 6}], [{'crosstie': False, 'idx': 7}]],
 [[{'crosstie': False, 'idx': 8}], [{'crosstie': False, 'idx': 9}]],
 [[{'crosstie': False, 'idx': 10}], [{'crosstie': False, 'idx': 11}]],
 [[{'crosstie': False, 'idx': 12}], [{'crosstie': False, 'idx': 13}]],
 [[{'crosstie': False, 'idx': 14}], [{'crosstie': False, 'idx': 15}]],
 [[{'crosstie': False, 'idx': 16}], [{'crosstie': False, 'idx': 17}]],
 [[{'crosstie': False, 'idx': 18}], [{'crosstie': False, 'idx': 19}]],
 [[{'crosstie': False, 'idx': 20}], [{'crosstie': False, 'idx': 21}]],
 [[{'crosstie': False, 'idx': 22}], [{'crosstie': False, 'idx': 23}]]]
```
which is a 12(substring) x 2(column per substring) x 1(row per column) module
[image]

You don't mention bypass diodes, and I believe your "series parallel" or "SP" module
>connected in series parallel (SP) as 4 strings of 24 cells

I think this is what PVMismatch calls a _partially_ cross tied:
[image]
pardon my poor artwork

mikofski, 2020-05-02:
Looking through the rest, I made a few changes, but unfortunately, I think I get the same result as what you originally got:

<details>

```python

from matplotlib import pyplot as plt
import numpy as np
from pvmismatch.pvmismatch_lib import (pvcell, pvconstants, pvmodule,
                                       pvstring, pvsystem)

plt.ion()

str_len = 1 # number of modules in a string
str_num_SP = 4 # number of (parallel connected) strings in the array
str_num_TCT = 1 # number of (parallel connected) strings in the array
v_bypass = -0.3  # [V] trigger voltage of bypass diode
cell_area = 136.5*65/312
Isc0_T0 = 4.7*65/312 # [A] reference short circuit current Mia Sole http://miasole.com/wp-content/uploads/2014/09/SolarCell_Datasheet_5.pdf
Isat1_T0 = 7.3E-11 # [A] ~ diode one saturation current see: https://www.sciencedirect.com/science/article/pii/S0927024811001565  fig 7.
Isat2_T0 = 4.8601e-05 # [A] ~ diode two saturation current see: https://www.sciencedirect.com/science/article/pii/S0927024811001565 fig 8.
Rs = 0.08 # [ohm] series resistance - based on data obtained from another intern
Rsh = 5 # [ohm] shunt resistance - based on data obtained from another intern
Eg = 1.05 # ~ CIGS bangap https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=4149613&tag=1
alpha_Isc = 0.0080

PVCONST = pvconstants.PVconstants(1001)

#1 used for series parallel system, 2 used for total cross tied system
pv_mod_pattern1 = pvmodule.crosstied_cellpos_pat([12, 12], 4, True)
pv_mod1 = pvmodule.PVmodule(
    cell_pos=pv_mod_pattern1, pvcells=None, pvconst=PVCONST,
    Vbypass=v_bypass, cellArea=cell_area)
pv_cells1 = pv_mod1.pvcells
for c1 in pv_cells1:
    c1.update(
        Isc0_T0=Isc0_T0, Isat1_T0=Isat1_T0, Isat2_T0=Isat1_T0, Eg=Eg,
        Rs=Rs, Rsh=Rsh)

pv_mod_pattern2 = pvmodule.crosstied_cellpos_pat([12, 12], 4, False)
pv_mod2 = pvmodule.PVmodule(
    cell_pos=pv_mod_pattern2, pvcells=None, pvconst=PVCONST,
    Vbypass=v_bypass, cellArea=cell_area)
pv_cells2 = pv_mod2.pvcells
for c2 in pv_cells2:
    c2.update(
        Isc0_T0=Isc0_T0, Isat1_T0=Isat1_T0, Isat2_T0=Isat1_T0, Eg=Eg,
        Rs=Rs, Rsh=Rsh)

f, ax = plt.subplots(2, 2, figsize=(16, 16))

#plt.figure(figsize=(8,8))
ax[0, 0].set_title('hor12b - TCT')
ax[0, 0].plot(pv_mod2.Vmod, pv_mod2.Imod, '', color='c', label='Simulated')
ax[0, 0].plot(pv_mod2.Vmod, pv_mod2.Pmod, '--', color='c')
ax[0, 0].set_xlim(0, 20)
ax[0, 0].set_ylim(0, 330)
ax[0, 0].set_xlabel('Voltage [V]')
ax[0, 0].set_ylabel('Current [I] / Power [W]')
ax[0, 0].legend()
ax[0, 0].grid()
    
#plt.figure(figsize=(8,8)) 
ax[0, 1].set_title('hor12b - SP')
ax[0, 1].plot(pv_mod1.Vmod, pv_mod1.Imod, '', color='c', label='Simulated')
ax[0, 1].plot(pv_mod1.Vmod, pv_mod1.Pmod, '--', color='c')
ax[0, 1].set_xlim(0, 20)
ax[0, 1].set_ylim(0, 330)
ax[0, 1].set_xlabel('Voltage [V]')
ax[0, 1].set_ylabel('Current [I] / Power [W]')
ax[0, 1].legend()
ax[0, 1].grid()

G = np.ones((24,4))*0.875
G[12:24,0:4] = 0.2
pv_mod1.setSuns(G.T.flatten())
pv_mod2.setSuns(G.T.flatten())

#plt.figure(figsize=(8,8))
ax[1, 0].set_title('hor12b - TCT')
ax[1, 0].plot(pv_mod2.Vmod, pv_mod2.Imod, '', color='c', label='Simulated')
ax[1, 0].plot(pv_mod2.Vmod, pv_mod2.Pmod, '--', color='c')
ax[1, 0].set_xlim(0, 20)
ax[1, 0].set_ylim(0, 20)
ax[1, 0].set_xlabel('Voltage [V]')
ax[1, 0].set_ylabel('Current [I] / Power [W]')
ax[1, 0].legend()
ax[1, 0].grid()
    
#plt.figure(figsize=(8,8)) 
ax[1, 1].set_title('hor12b - SP')
ax[1, 1].plot(pv_mod1.Vmod, pv_mod1.Imod, '', color='c', label='Simulated')
ax[1, 1].plot(pv_mod1.Vmod, pv_mod1.Pmod, '--', color='c')
ax[1, 1].set_xlim(0, 20)
ax[1, 1].set_ylim(0, 20)
ax[1, 1].set_xlabel('Voltage [V]')
ax[1, 1].set_ylabel('Current [I] / Power [W]')
ax[1, 1].legend()
ax[1, 1].grid()

f.tight_layout()
```
</details>
[image]

mikofski, 2020-05-03:
I've been thinking about this, and I had the following thoughts.

1. due to the shading pattern, but the PCT and TCT modules should have identical response. 
2. a simple analog to both modules would be a 2-cells-in-series model with one cell shaded.
3. To me it looks like PVMismatch is modeling this 2-cells-in-series simple analog correctly
4. The experimental measurements do not seem consistent with the shading on the lower half of the module, which you give as 0.2-suns. The measurements approach nearly zero current which would indicate that there was also nearly zero light on the lower half of the module.

So repeating the simulation with more shading on the bottom half of the modules, `Ee=0.02` suns makes plots that seem closer to the experimental measurements
[image]

5. the last thing I noticed is that the reverse bias characteristics are never defined, but maybe it doesn't matter. I don't know what they would be for CIGS, but when I vary them it doesn't affect the shape of the curves anyway, so I guess thy bypass diodes are doing the right thing.

Also I noticed that the `shade_query.txt` snippet updated the cell characteristics, but it's probably easier to set them when instantiating the cell, and then pass the cell to each module. Be careful to call `setSuns` or make a copy so the 2 modules reference different objects. In this examples `setSuns` does the job of making copies of the cell object references so it's okay.

<details>

```python
from matplotlib import pyplot as plt
import numpy as np
from pvmismatch import *

plt.ion()

v_bypass = -0.3  # [V] trigger voltage of bypass diode
cell_area = 136.5*65/312
Isc0_T0 = 4.7*65/312 # [A] reference short circuit current Mia Sole http://miasole.com/wp-content/uploads/2014/09/SolarCell_Datasheet_5.pdf
Isat1_T0 = 7.3E-11 # [A] ~ diode one saturation current see: https://www.sciencedirect.com/science/article/pii/S0927024811001565  fig 7.
Isat2_T0 = 4.8601e-05 # [A] ~ diode two saturation current see: https://www.sciencedirect.com/science/article/pii/S0927024811001565 fig 8.
Rs = 0.08 # [ohm] series resistance - based on data obtained from another intern
Rsh = 5 # [ohm] shunt resistance - based on data obtained from another intern
Eg = 1.05 # ~ CIGS bangap https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=4149613&tag=1
alpha_Isc = 0.0080

PVCONST = pvconstants.PVconstants(1001)

pv_cell = pvcell.PVcell(
    Isc0_T0=Isc0_T0, Isat1_T0=Isat1_T0, Isat2_T0=Isat1_T0, Eg=Eg,
    Rs=Rs, Rsh=Rsh, alpha_Isc=alpha_Isc, pvconst=PVCONST, VRBD=-10.0)

#1 used for series parallel system, 2 used for total cross tied system
pv_mod_pattern1 = pvmodule.crosstied_cellpos_pat([12, 12], 4, True)
pv_mod1 = pvmodule.PVmodule(
    cell_pos=pv_mod_pattern1, pvcells=pv_cell,
    Vbypass=v_bypass, cellArea=cell_area)

pv_mod_pattern2 = pvmodule.crosstied_cellpos_pat([12, 12], 4, False)
pv_mod2 = pvmodule.PVmodule(
    cell_pos=pv_mod_pattern2, pvcells=pv_cell,
    Vbypass=v_bypass, cellArea=cell_area)

f, ax = plt.subplots(2, 1, figsize=(8, 8))

G = np.ones((24,4))*0.875
G[12:24,0:4] = 0.02  # more shade on bottom cells

pv_mod1.setSuns(G.T.flatten())
pv_mod2.setSuns(G.T.flatten())

#plt.figure(figsize=(8,8))
ax[0].set_title('hor12b - TCT')
ax[0].plot(pv_mod2.Vmod, pv_mod2.Imod, '', color='c', label='Simulated')
ax[0].plot(pv_mod2.Vmod, pv_mod2.Pmod, '--', color='c')
ax[0].set_xlim(0, 15)
ax[0].set_ylim(0, 20)
ax[0].set_xlabel('Voltage [V]')
ax[0].set_ylabel('Current [I] / Power [W]')
ax[0].legend()
ax[0].grid()
    
#plt.figure(figsize=(8,8)) 
ax[1].set_title('hor12b - SP')
ax[1].plot(pv_mod1.Vmod, pv_mod1.Imod, '', color='c', label='Simulated')
ax[1].plot(pv_mod1.Vmod, pv_mod1.Pmod, '--', color='c')
ax[1].set_xlim(0, 15)
ax[1].set_ylim(0, 20)
ax[1].set_xlabel('Voltage [V]')
ax[1].set_ylabel('Current [I] / Power [W]')
ax[1].legend()
ax[1].grid()

f.tight_layout()
```
</details>

a-karanth, 2020-05-08:
Hi @mikofski,
Thank you for your response and the suggestions you made.
I did not mention in my original note that bypass diodes are connected across every two cells since they are CIGS cells. So that makes it - one bypass diode across every two cells in each string in the series parallel connection, and one bypass diode across every two rows in the total cross tight connection. You initial representation in the table is indeed how the substrings are defined (in one string of SP). 

I did however change the irradiance definition on the shaded cells and also correct the typo that you pointed out. And I am happy to say that I have gotten results much closer to the experimental values. I also simulated more shading patterns which albeit slightly different, are much closer to the experimental results than I was getting before. I have attached a couple below in case you are interested. Again, I sincerely appreciate and thank you for your help.

[image]
[image]

mikofski, 2020-05-09:
@a-karanth Your new results look great! There are some small differences still so there may still be room for improvement, but still very impressive. Do you plan to publish these results? Also if you're concerns/questions have been satisfied, then you might consider closing this issue. Thanks!

a-karanth, 2020-05-13:
Thank you. I am working on this for my thesis which will be publicly available in the TU Eindhoven library. As for publication, I am not yet sure about that.

closed 2020-05-13 by a-karanth

lp74, 2021-03-01:
Hu @mikofski, 

would it be possible to have help with this standard configuration? I am not sure.

[image]

mikofski, 2021-03-01:
Hi @lp74 , This question might be better for GitHub's new [discussions](https://github.com/SunPower/PVMismatch/discussions) feature. heck out the [Q&A discussion](https://github.com/SunPower/PVMismatch/discussions/145) I started on this topic. Let me know over there if my comments help or not. Thanks!

lp74, 2021-03-02:
thank you very much