SunPower/PVMismatch #129: PVcell.calcVcell and PVmodule.calcCell do not return the same results

akarakishKWS, 2020-06-25:
Hello all!

I've noticed that the `calcVcell` method from the PVCell class does not yield the same result as the `calcCell` method from the PVModule class.

Here is the result I get from the following snippet of code:

[image]

```python 
import numpy as np
import pvmismatch as pvm
from matplotlib import pyplot as plt

# Initialize PVcell @ 250 W/m²
pvcell_250 = pvm.pvcell.PVcell(Ee=0.25)

# Initialize PVmodule @ 250 W/m²
STD72 = pvm.pvmodule.standard_cellpos_pat(12, [2, 2, 2])
pvmodule_250W = pvm.pvmodule.PVmodule(
    cell_pos=STD72,
    pvcells=pvcell_250
)

# IV with PVcell.calcVcell()
I_cell = np.arange(0, 6.5, 0.01)
V_cell = [pvcell_250.calcVcell(x) for x in I_cell]

# PVmodule.calcCell()
Icell_mod, Vcell_mod, _ = map(lambda x: x.flatten(), pvmodule_250W.pvcells[1].calcCell())

# Plot IV curves
plt.figure(figsize=(12, 8))
plt.scatter(V_cell, I_cell, c='blue', label='PVcell.calcVcell()')
plt.scatter(Vcell_mod, Icell_mod, c='red', label='PVmodule.calcCell()')
plt.xlim(-7, 1)
plt.ylim(0, 4)
plt.legend()
plt.savefig("error.png")
plt.show()
```

As you can see, I think we should expect the two figures to be identitical.

Has anyone encountered the same issue?

Thanks!

mikofski, 2020-06-26:
@akarakishKWS thanks for using PVM. The search functions in pvcell (`calcVcell` & `calcIcell`) don't include the reverse bias breakdown current, because they're only used to find Voc & Isc exactly when generating coefficients:

https://github.com/SunPower/PVMismatch/blob/e733ebbc8ada517f36da9bccf6569f6e3a55fb4c/pvmismatch/contrib/gen_coeffs/example.py#L79

https://github.com/SunPower/PVMismatch/blob/e733ebbc8ada517f36da9bccf6569f6e3a55fb4c/pvmismatch/contrib/gen_coeffs/example.py#L83

You should use [`pvcell.calcCell`](https://sunpower.github.io/PVMismatch/api/pvcell.html#pvmismatch.pvmismatch_lib.pvcell.PVcell.calcCell) which uses Bishop's explicit method to return the entire IV curve including reverse bias in one call. Only use the search functions to find a single specific point in the _forward bias_ region of the curve (quadrant I) when the _full_ curve isn't needed.

What is your use case? PVM is always looking for improvements. I'm happy to review your contributions. If you think there's a strong case for adding the reverse bias term to the search functions, I'm open to it. 

Also PTAL at the [quickstart](https://sunpower.github.io/PVMismatch/tutorials/quickstart.html), maybe there are some improvements that could be made to the documentation that would address your use cases batter?

Thanks!

akarakishKWS, 2020-06-30:
@mikofski thanks for your prompt reply. 
I understand now, I only will use ```pvcell.calcCell```.

I was doing some tests on different configurations, to see if I'll get the same results as the main methods. I started from a 'cell' level tests to 'pv system' level tests.

Thanks again!

mikofski, 2020-06-30:
@akarakishKWS sounds good. I can't wait to see the results of your tests. If you're happy with this answer can you please close this issue? Thanks!

akarakishKWS, 2020-07-03:
@mikofski Great ! Thanks again for your prompt replies.

closed 2020-07-03 by akarakishKWS