SunPower/PVMismatch #130: .PAN file parameters

akarakishKWS, 2020-07-10:
Hello all, 

Has anyone tried to initialize a PVcell using the paramaters from a .PAN file ? 
I am having an issue with that matter.

Thanks !

mikofski, 2020-07-10:
Sorry, it will be hard. You will need to regenerate them using `gen_coeffs` in `pvm.contrib` after using pvlib or pvsyst to output a matrix of iv curves. That would be a great contribution of your interested. If you look at `contrib`, I think there's already an example that uses the Sandia model (aka: `sapm`) to output iv curves and then generates coefficients for PVM

cwhanse, 2020-07-10:
The parameters in a .PAN file are for PVsyst's single diode model for the IV curve, and are for a module, not a cell. PVMismatch uses a 2-diode model with different irradiance and temperature dependences. As @mikofski suggests, you could use the parameters in the .PAN file to calculate module IV curves (pvlib-python could help there), scale the IV curve down to a cell, then get parameters for PVMismatch's cell model from the scaled IV curve.

akarakishKWS, 2020-07-13:
Hello all, 

Thank you for your replies. 

At the beginning I tried ````get_coeffs``` but it doesn't always converge. For the same modules it converges for some classes and not the other (cf. any module datasheet with multiple nominal power).

However, I found a way to make it converge. The method ```get_coeffs``` uses an optimization algorithm and it takes the ```PVcell``` default coefficients as it's initial values. 

So, to make it converge when using a "module class" that doesn't converge with default initial values, I use the coefficients resulting from the optimization on the class that converges as initial values.  I hope my explanation is clear.

```
# Class 1 with default initial values: converges
Pmp = 400
Isc = 10.61
Voc = 49.1
Imp = 9.96
Vmp = 40.16
nb_cell = 72

x1, sol1 = gen_coeffs.gen_two_diode(isc=Isc,
                                    voc=Voc,
                                    imp=Imp,
                                    vmp=Vmp,
                                    nseries=nb_cell,
                                    nparallel=1,
                                    tc=25.0,
                                   )
print("sol1 : ", sol1.message)

# Class 2 with default initial values:  doesn't converge
Pmp2 = 380
Isc2 = 10.3
Voc2 = 48.2
Imp2 = 9.72
Vmp2 = 39.1

x2, sol2 = gen_coeffs.gen_two_diode(isc=Isc2,
                                    voc=Voc2,
                                    imp=Imp2,
                                    vmp=Vmp2,
                                    nseries=nb_cell,
                                    nparallel=1,
                                    tc=25.0,
                                   )
print("sol2 : ", sol2.message)

# Class 2 with class 1 results as initial values:  converges
x3, sol3 = gen_coeffs.gen_two_diode(isc=Isc2,
                                    voc=Voc2,
                                    imp=Imp2,
                                    vmp=Vmp2,
                                    nseries=nb_cell,
                                    nparallel=1,
                                    tc=25.0,
                                    x0=x1,
                                   )
print("sol3 : ", sol3.message)
```

> sol1 :  The solution converged.
> sol2 :  The iteration is not making good progress, as measured by the improvement from the last ten iterations.
> sol3 :  The solution converged.

Here's my pvmismatch setup

```
Name: pvmismatch
Version: 4.1
Summary: PV Mismatch Calculator
```

Thanks!

mikofski, 2020-07-13:
This does make a lot of sense, and is exactly what I would recommend. The solver is the SciPy Newton implementation which is prone to issues if it is too far from the final solution, so it's a Catch-22 of sorts. There may be better ways to estimate the initial guess, but your solution seems to work for now. If you're satisfied, can you close this issue? thanks!

closed 2020-07-15 by akarakishKWS