SunPower/PVMismatch #132: Some rookie issues; need help figuring out the documentation

abooda1981, 2020-10-03:
Hello; thanks for this wonderful library, it's incredibly useful from what I can see. Thanks especially to @mikofski for suggesting I look into it for my issue trying to work with partial shading and then calculate the heating. 

I've been building "randomised" (it's a bit more complicated than that, but that's another story) shading patterns which are in the shape of a 12 X 6 grid (a data frame; 12 strings of 6 cells each) in R which I want to then feed into pvmismatch in order to obtain the cell powers (please tell me if I'm barking up the wrong tree at this point). I simply am not sure how to implement this at the moment. Going on what I've seen in the tutorial, it would be something like: 

``
pvsys5.setSuns(0:{first vector}, 1:{second vector} ...and so on)
``
``
pvsys5.Psys
``
...my first assumption was that this was the power generated by each cell, but I guess not.

mikofski, 2020-10-03:
Thanks for trying PVMismatch. I agree, the documentation needs improvement. `Psys` is an undocumented attribute of `pvsystem.PVsystem` that gives you the system power as a function of `Vsys`. Use `ploySys()` to see the full power vs voltage curve. Can you create an issue describing the need to add these attributes to the docs?

Can you provide more info on your use case? I know it has to do with hotspots. Are you trying to make a single PV module, a string of them, or an entire PV system?

abooda1981, 2020-10-04:
Thanks, I will do as you suggest and open an issue about the documentation and the need for descriptions about the Psys and so on. 

With regards to my usage case: I want to model a single module of 72 cells in total, which are divided into 6 strings/substrings of 12 cells each. It builds on the following more or less: https://www.mdpi.com/1996-1073/11/1/250 (although this paper uses a 60-cell module). 

Looking through the older issues, I have found [this](https://github.com/SunPower/PVMismatch/issues/122) to be quite similar:  I know what kind of pattern I want for the diodes (TCT), and what the layout of the cells should be within the module. Simply put though, I am unsure about how to translate that to the syntax required for pvmismatch. My best guess is something like this: 

``
my_pattern = pvmodule.standard_cellpos_pat(ncols_per_substr=[12], nrows=6)
my_module = pvmodule.PVmodule(cell_pos=my_pattern, pvcells=None, pvconst=None, Vbypass=-0.3, cellArea=0.024)
``
I'd be grateful if people could point out whether or not the above flies. 

Thanks a million! For what it's worth, once I submit my work in early December I would be more than happy to contribute to a kind of "novice's guide" or something aimed at less technical users. I think there is a huge user base for this library who might otherwise be daunted by the documentation and I'd love to help.

abooda1981, 2020-10-05:
Hello. I think I ought to close this as I have been looking through some of the documentation and what I'm looking for does seem to be in there, just deeply buried. 

Thanks again.

closed 2020-10-05 by abooda1981