SunPower/PVMismatch #134: Trying to iterate a module to create a list of photogenerated currents

abooda1981, 2020-10-06:
Hello all. I hope nobody minds my using this space to raise an issue. I want to automate the process of calculating the power losses due to partial shading. As a stepping stone, I have the code below, which I think should be fairly self explanatory. It copies a single module; makes a fairly arbitrary shading pattern, and then places the value of the photogenerated current in the appropriate index of a list. 

```
#Create a grid with some arbitrarily shaded bits
#Defined as a proportion of "suns"
shades = np.ones((10,6))
shades[4:8, 0:6] = 0.323
my_module_copy = my_module_shaded
#Define a function which takes the shading pattern and applies each one to the relevant cell
#in a module
def generate_photo_currents(fed_shades, fed_module):
  k: int = 0
  generated_photo_currents = []
  for i in range(0, 10):
    for j in range(0, 6):
     fed_module.setSuns(fed_shades[i,j], k)
     generated_photo_currents.append(fed_module.pvcells[k].Igen)
     k = k + 1
     return generated_photo_currents
```

I then tried to apply the above function and I received the following error in the TraceBack:

```
made_currents = generate_photo_currents(shades, my_module_copy)

Traceback (most recent call last):
  File ".../PyCharmCE2020.2/scratches/scratch_2.py", line 238, in <module>
    made_currents = generate_photo_currents(shades, my_module_copy)
  File ".../PyCharmCE2020.2/scratches/scratch_2.py", line 233, in generate_photo_currents
    fed_module.setSuns(fed_shades[i,j], k)
  File ".../python3.7/site-packages/pvmismatch/pvmismatch_lib/pvmodule.py", line 323, in setSuns
    cells_to_update = [self.pvcells[i] for i in cells]
TypeError: 'int' object is not iterable
``` 
I can't see why this code is failing. Any help would be greatly appreciated.

adambgnr, 2020-10-06:
Hi @abooda1981 ,

Setting the cell irradiance one by one can be a bit tricky in PVMismatch. Instead, I recommend to set the irradiance values for the entire module in one go, like this:

```
import pvmismatch
import numpy as np
import pandas as pd

#Create a grid with some arbitrarily shaded bits
#Defined as a proportion of "suns"
shades = np.ones((10,6))
shades[4:8, 0:6] = 0.323

# I guess your PV module looks like this:
pv_mod_pattern = pvmismatch.pvmodule.standard_cellpos_pat(nrows=10, ncols_per_substr=[2]*3)
my_module_copy = pvmismatch.pvmodule.PVmodule(cell_pos=pv_mod_pattern)
fed_shades = shades
fed_module = my_module_copy

# define a function that returns the cell positions in a dataframe
def create_cell_pos_df(pv_mod):
    """Create cell position dataframe of a module"""
    cell_pos = pv_mod.cell_pos
    nrows = int(pv_mod.numberCells / sum(pv_mod.subStrCells))
    cell_pos_df = pd.DataFrame(index=['{}'.format(nr)
                                      for nr
                                      in range(nrows)])
    for b, bypass in enumerate(cell_pos):
        for c, col in enumerate(bypass):
            cell_pos_df['{}_{}'.format(b, c)] = [i['idx'] for i in col]
    return cell_pos_df

# make the cell position DataFrame
cell_pos = create_cell_pos_df(pv_mod=fed_module)

# the indexes in the DataFrame are the cell row numbers
# the column names are cell substring number and column number within cell
# substring separated with a "_"
print(cell_pos)

# convert it to a np array
cell_pos_np = cell_pos.to_numpy()

# now it is a np array that has the same shape as the shade matrix
# and the location of the shade is in the right nplace:
print(cell_pos_np)
print(fed_shades)

# now we can set the suns in one go for the whole PV module
fed_module.setSuns(fed_shades, cell_pos)

# and check if it is indeed set
print(fed_module.Ee)

```
Btw PVMismatch has a contrib, wich with you can draw the shading and temperature patterns for your PV system in excel and import them to simulate in PVMismatch (I adapted the above code from there). You can find it here:
https://github.com/SunPower/PVMismatch/tree/master/pvmismatch/contrib/xlsio 

And there is an other contrib, which with you can set the shades for a standalone PV module and plot the I-V curves. This one even has a nice GUI! You can find it here:
https://github.com/SunPower/PVMismatch/blob/master/pvmismatch/contrib/module_mismatch_simulator.py

abooda1981, 2020-10-07:
Hello, and thanks @adambgnr ! Sorry I ought to have gotten back yesterday but got caught up in something else. Your code suggestion does indeed work although it is a bit more complex than expected. 

I will close this now as your solution works.

closed 2020-10-07 by abooda1981