SunPower/PVMismatch #135: create a google group, discourse, or other forum for users to ask questions

mikofski, 2020-10-09:
Currently if users have a usage question their only option is to create an issue. This creates a barrier to free exchange of ideas, as new users may feel their questions don't warrant an issue. A google group, discourse, or php forum/bulletin board would alleviate this.

kanderso-nrel, 2020-10-09:
FYI: pvmismatch has a small but nonzero presence on stackoverflow: https://stackoverflow.com/search?q=pvmismatch

No comment from me on whether that's better or worse than a google group.

chetan201, 2020-11-06:
that's really great. 
I will add a note regarding that on the landing page and request users to tag pvmismatch.

abooda1981, 2020-11-22:
A Google users group would be awesome!

chetan201, 2020-12-09:
@mikofski I started a GitHub discussions for this repo. Do you think it addresses this issue?

closed 2020-12-09 by mikofski