SunPower/PVMismatch #139: Question about gen_coeffs.gen_two_diode

PyroStadlin, 2020-12-20:
I wanted to ask you a last question regarding PVMismatch. I do not really understand the Output of gen_coeffs.gen_two_diode, which is:
 
```
fjac: array([[nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan,nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan,
... nan, nan, nan, nan, nan, nan, nan, nan]])
     fun: array([-5.00391537e-02, -5.32664899e-02, -5.71223793e-02, -  5.97740265e-02,-6.18808152e-02, -6.36665900e-02...-1.94677609e-07,
-2.92041963e-07, -3.89423353e-07, -4.86821782e-07, -5.35527386e-07])
    ipvt: array([1, 2, 3, 4], dtype=int32)
message: 'The cosine of the angle between func(x) and any column of the\n  Jacobian is at most 0.000000 in absolute value'
  nfev: 1
  njev: 1
  qtf: array([nan, nan, nan, nan])
  status: 4
success: True
x: array([-24.51750581, -13.41004545,   0.06324555,   3.16227766])
```
Can I find the values which I need (Isat1, Isat2, Rs and Rs) in this output ?  or  after using the gen_coeffs.gen_two_diode where can I find the parameters needed for the 2 diode model ?

chetan201, 2021-01-14:
@PyroStadlin 
gen_coeffs returns this tuple -> (isat1, isat2, rs, rsh), sol
which is what you are looking for to use the 2 diode model.

As for the non-convergence, as I communicated over email - it's most likely an issue of Python, Numpy or Scipy package version mismatch. Please try matching the requirements I shared or the ones in the repo.

Also, I'd encourage you to please move this issue to Discussions section since it's not established that it's a bug yet. Thanks!

closed 2021-01-14 by chetan201

mikofski, 2021-01-16:
Please see [discussion](https://github.com/SunPower/PVMismatch/discussions/143) https://github.com/SunPower/PVMismatch/discussions/143