SunPower/PVMismatch #140: Fix failing builds: python 2.7 is failing on travis CI & TravisCI.org is about to stop service anyway

mikofski, 2021-01-13:
Here's how it looks in the docs and README:
[image]

Follow the link: https://travis-ci.org/github/SunPower/PVMismatch, and you see it's the Python-2.7 build that fails. Has PVM officially dropped Py2?

And anyway, there's this alert from TravisCI.org:
[image]

So that means that builds and testing need to be migrated to Azure pipelines. We can use what pvlib has done to figure it out

chetan201, 2021-01-14:
@mikofski I took a look and couldn't find any obvious reasons why Py27 should fail. If Azure pipelines is easy to migrate to, I also support the idea. What do you think about Github actions? 
A cursory search landed me here - https://docs.github.com/en/free-pro-team@latest/actions/guides/about-continuous-integration

kanderso-nrel, 2021-01-14:
Looks like the `dulwich` dependency is the culprit.  Per https://www.dulwich.io/docs/#supported-versions-of-python, they dropped support for python 2.7 starting in `0.20`.  The [failed build](https://travis-ci.org/github/SunPower/PVMismatch/jobs/741997843#L385) installs `0.20.2` while the [last passing build](https://travis-ci.org/github/SunPower/PVMismatch/jobs/689364938#L460) installed `0.19.16`.  So I guess a small update to `requirements.txt` would get the builds passing again.