SunPower/PVMismatch #158: Different cell irradiances

AugustVan, 2024-04-04:
I saw in a previous "issue #107 "  that @mikofski made following code to simulate a PV panel with different shades: https://github.com/SunPower/PVMismatch/blob/master/pvmismatch/contrib/module_mismatch_simulator.py.
I am also trying to simulate something simular. The difference in my project is that I already have the different irradiances for each cell of the PV panel. I got these from an irradiance model I made with bifacial radiance from NREL (with help of Silvana). My question now is if it is possible to give each cell a different irradiance value? In the code mikofski wrote, the values were the same for all the cells. So I am not sure how to change this and if this is even possible?

mikofski, 2024-04-04:
Use [`PVsystem.setSuns()`](https://sunpower.github.io/PVMismatch/api/pvsystem.html#pvmismatch.pvmismatch_lib.pvsystem.PVsystem.setSuns) method to individually set every cell irradiance.  

Check out the example on [quickstart](https://sunpower.github.io/PVMismatch/tutorials/quickstart.html)

AugustVan, 2024-04-04:
So the module I use has 132 cells (shown in picture below) and with this code I should be able to give every cell a unique irradiance?
Ee={0: {0: {'cells': (...), 'Ee': (...)}}}
pvsys.setSuns(Ee)

And if what I say is correct, is this code also correct to give me the IV-curve and PV-curve of the module that I'm trying to simulate?
"
import pvmismatch as pvm 
halfcell132_bypass3 = pvm.pvmodule.crosstied_cellpos_pat([22, 22, 22], 2, partial=True)
pvmod = halfcell132_bypass3
pvsys = pvm.PVsystem(numberStrs=1, numberMods=1, pvmods=pvmod)  
Ee={0: {0: {'cells': (...), 'Ee': (...)}}}
pvsys.setSuns(Ee)
print ("Pmp: %f [W], Eff: %f [%%], FF: %f [%%]" % (pvsys.Pmp, pvsys.eff * 100., pvsys.FF * 100.))
print ("Imp: %f [A], Vmp: %f [V], Isc: %f [A], Voc: %f [V]" % (pvsys.Imp, pvsys.Vmp, pvsys.Isc, pvsys.Voc))
PowerDetailed=pvsys.Pmp  
pvsys.plotSys()      
"
    
[image]

Kind regards

August

AugustVan, 2024-04-05:
So this is the code I have written but I don't know if it's wrong or if something is missing because the output power I get is pretty low. I'm working with a 560W panel and these are the curves I get back. 
[image]
The code I wrote in bifacial radiance measures the irradiance every hour of the day from 5:00 till 20:00. The irradiance values I used here are from 16:00, the average irradiance is 0,67 so the output value can't be so low. Is there a way to how I can fix this? 
## code:
import pvmismatch as pvm # this imports everything we need
import numpy as np
import csv
numcells = 132
portraitorlandscape = 'landscape '

if numcells == 132:
    stdpl=np.array(
    [[  0,  22,  44,  66,  88, 110],
     [  1,  23,  45,  67,  89, 111],
     [  2,  24,  46,  68,  90, 112],
     [  3,  25,  47,  69,  91, 113],
     [  4,  26,  48,  70,  92, 114],
     [  5,  27,  49,  71,  93, 115],
     [  6,  28,  50,  72,  94, 116],
     [  7,  29,  51,  73,  95, 117],
     [  8,  30,  52,  74,  96, 118],
     [  9,  31,  53,  75,  97, 119],
     [ 10,  32,  54,  76,  98, 120],
     [ 11,  33,  55,  77,  99, 121],
     [ 12,  34,  56,  78, 100, 122],
     [ 13,  35,  57,  79, 101, 123],
     [ 14,  36,  58,  80, 102, 124],
     [ 15,  37,  59,  81, 103, 125],
     [ 16,  38,  60,  82, 104, 126],
     [ 17,  39,  61,  83, 105, 127],
     [ 18,  40,  62,  84, 106, 128],
     [ 19,  41,  63,  85, 107, 129],
     [ 20,  42,  64,  86, 108, 130],
     [ 21,  43,  65,  87, 109, 131]])
   
if portraitorlandscape == 'landscape':
    stdpl = stdpl.transpose()

cellsx = len(stdpl[1]); cellsy = len(stdpl)

if cellsx*cellsy == 132:
    cell_pos = pvm.pvmodule.STD132

pvmod=pvm.pvmodule.PVmodule(cell_pos=cell_pos)

pvsys_front = pvm.PVsystem(numberStrs=1, numberMods=1, pvmods=pvmod)   

array = np.arange(132)
print(array)

with open('C:/Users/Gebruiker/Documents/school/iiw4_MA1/bifacial_radiance/TEMP/August/results/irr_1axis_2021-06-01_1600Baseline_Row2_Module3.csv', newline='') as csvfile:
    reader = csv.DictReader(csvfile)
    Wm2Front = 'Wm2Front'
    Wm2Front = [float(row[Wm2Front]) / 1000 for row in reader]
print(Wm2Front)

Ee={0: {0: {'cells': array, 'Ee': Wm2Front}}}
pvsys_front.setSuns(Ee)
print ("Pmp: %f [W], Eff: %f [%%], FF: %f [%%]" % (pvsys_front.Pmp, pvsys_front.eff * 100., pvsys_front.FF * 100.))
print ("Imp: %f [A], Vmp: %f [V], Isc: %f [A], Voc: %f [V]" % (pvsys_front.Imp, pvsys_front.Vmp, pvsys_front.Isc, pvsys_front.Voc))
PowerDetailed=pvsys_front.Pmp  
pvsys_front.plotSys() 
## end of code:

Kind regards

August

mikofski, 2024-04-05:
The default 2-diode model parameters are for a 96 cell 315W SunPower E20 module. Did you update the cell properties? `Isat, Rs, Rsh` etc?

AugustVan, 2024-04-06:
No I haven't, where could I find these so I can change them?

AugustVan, 2024-04-06:
Think I found the code:
https://github.com/SunPower/PVMismatch/blob/master/pvmismatch/contrib/gen_coeffs/__init__.py#L27
Is this it?

AugustVan, 2024-04-06:
No I haven't, I think I found where I can change these

AugustVan, 2024-04-06:
Probably a dumb question (I'm not the biggest expert in this) but how can I calculate the Isat, Rs, Rsh, etc. Is there a site you can give me where I can calculate these with? this is the data sheet I use and the only things I have given are the: Vmp, Imp, Voc and the Isc. Does this suffice to calculate the parameters? Again sorry for these "basic" questions as I'm not the greatest in these calculations.
[DT-M  Datasheet_Vertex_DEG19RC.20_EN_2022 565-575Wp (1).pdf](https://github.com/SunPower/PVMismatch/files/14893525/DT-M.Datasheet_Vertex_DEG19RC.20_EN_2022.565-575Wp.1.pdf)

Kind regards

August

mikofski, 2024-04-06:
Yes, you are correct. Use the values you find in data sheet and put them into https://sunpower.github.io/PVMismatch/api/contrib.html#pvmismatch.contrib.gen_coeffs.gen_two_diode

In addition to `Isc, Imp, Voc,Vmp` you also need `Nseries=66, Nparallel=2, alpha_Isc, beta_Voc`
There are no dumb questions only dumb answers, and generating the coefficients is not trivial although it can be done. There aren’t good examples in the documentation, but there are a few scattered around In discussions and stack overflow. I will try to find and repost them.

mikofski, 2024-04-06:
This discussion: https://github.com/SunPower/PVMismatch/discussions/156 describes generating 2-diode model coefficients for a Trina module. I hope it helps

mikofski, 2024-04-06:
@markcampanelli maybe you can help?

markcampanelli, 2024-04-06:
> @markcampanelli maybe you can help?

I have had significant issues fitting 2-diode models consistently. Sometimes convergence is robust, and other times the parameters converge in a way that essentially reduces the model to a single-diode model. (Note: My efforts usually involved also fitting the two ideality factors instead of fixing n_1=1 and n_2=2, in which case n_2->infinity in some cases during the fit. Furthermore, this could be a "local" fit to a single dense I-V curve, or a "global" fit to an IEC 61853-1 matrix of sparse I-V data.)

Another question I would have here is how much do the parameter choices for the reverse breakdown of the cell diodes' and shunting by the bypass diodes' matter?

mikofski, 2024-04-06:
Good point, the default breakdown (`VRBD`) is only 5V, but I suspect it will be a bit larger for PERC, hopefully it won’t matter? But to be safe should experiment with different values like 10V or more

AugustVan, 2024-04-07:
When I try to run the example of gen coef I get this error: "from __future__ imports must occur at the beginning of the file". But as you can see there is no code above so it already is at the beginning of the file. Am I missing something here?

[image]

markcampanelli, 2024-04-07:
@AugustVan Curious what Python version you are on?

mikofski, 2024-04-07:
That example might be a little out of date, I bet you can just comment out that line, sorry, the`contrib` folder is not maintained

AugustVan, 2024-04-08:
> @AugustVan Curious what Python version you are on?

I'm currently on python 3.11

AugustVan, 2024-04-12:
> That example might be a little out of date, I bet you can just comment out that line, sorry, the`contrib` folder is not maintained

So I have tried "gencoef" in an older version python (python 3.7) and I got the same error. Then I commented that line out as you said but then I got an assertion error on line 40. After looking in to the error I found that "isc0" (that was calculated on line 39 in the example) was equel to "0.006429300000000002". This is one thousandth smaller than the ISC0 ("Isco": 6.4293) from the module "SunPower_SPR_E20_435". So because the difference between these two was so big this code "assert np.isclose(isc0, ISC0)" gave an assertion error. To "solve" I added this line of code "isc0 = isc0*1000" in the function "gen_sapm". I don't know if this solved the problem because the values on the graph itself are very small now but the values Rsh, Rs, Isat1 and Isat2 are still the same.
[image]

cwhanse, 2024-04-12:
I would double check that the input irradiance is in W/m2, and hasn't been normalized by dividing by 1000 W/m2.

AugustVan, 2024-04-17:
When I put these parameters in the code "PVcell" and run my code I get following error:
I think this is because I didn't run de gencoef code myself, but given circumstances the code gencoef won't work on my device.
##
runfile('C:/Users/Gebruiker/Documents/school/iiw4_MA1/masterproef/python code/PVmismatch/Simulatie_Front.py', wdir='C:/Users/Gebruiker/Documents/school/iiw4_MA1/masterproef/python code/PVmismatch')
Traceback (most recent call last):

  File ~\anaconda3\Lib\site-packages\pvmismatch\pvmismatch_lib\pvmodule.py:213 in __init__
    pvc0 = pvcells[0]

TypeError: 'NoneType' object is not subscriptable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File ~\anaconda3\Lib\site-packages\pvmismatch\pvmismatch_lib\pvmodule.py:217 in __init__
    pvconst = pvcells.pvconst

AttributeError: 'NoneType' object has no attribute 'pvconst'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File ~\anaconda3\Lib\site-packages\spyder_kernels\py3compat.py:356 in compat_exec
    exec(code, globals, locals)

  File c:\users\gebruiker\documents\school\iiw4_ma1\masterproef\python code\pvmismatch\simulatie_front.py:50
    pvmod=pvm.pvmodule.PVmodule(cell_pos=cell_pos)

  File ~\anaconda3\Lib\site-packages\pvmismatch\pvmismatch_lib\pvmodule.py:223 in __init__
    pvcells = PVcell(pvconst=pvconst)

  File ~\anaconda3\Lib\site-packages\pvmismatch\pvmismatch_lib\pvcell.py:58 in __init__
    self.Rs = Rs  #: [ohm] series resistance

  File ~\anaconda3\Lib\site-packages\pvmismatch\pvmismatch_lib\pvcell.py:92 in __setattr__
    super(PVcell, self).__setattr__(key, value)

TypeError: super(type, obj): obj must be an instance or subtype of type

mikofski, 2024-04-17:
Without your code I can’t help you troubleshoot, perhaps you can use a more stable computing environment like Google co-laboratory? Why can these coefficients be used for your device? They were generated from your spec sheet?

AugustVan, 2024-04-17:
https://gist.github.com/AugustVan/f59b77665c78419f716754fa5bdb592b, this is the gist I made but it doesn't find my file for some reason so i've put the file in attachment (with this file I try to simulate some shadows to see the effects of the bypass diodes, but at the moment these don't work). The problem with the coefficients you gave me is that when I type them in on this location:
[image]
I get the error I showed you above... I didn't change anything in the libraries, so I have no idea why I get those errors
[Map1.csv](https://github.com/SunPower/PVMismatch/files/15017201/Map1.csv)

mikofski, 2024-04-17:
Hooray! It works for me: https://gist.github.com/mikofski/4d2070a0e3da984a6c484c9b2147ff38.

I copied the Map1.csv values into the notebook, but to use in Colab, I think you would need to put them in your Google drive folder?

AugustVan, 2024-04-18:
Yes that worked for me, thank you so much for the help!
If I have some more questions I will ask them in a new issue.

closed 2024-04-18 by AugustVan